EmotiBit/EmotiBit_FeatherWing #34: Code Reorganization

badfishblues, 2019-07-15:
#### Software Considerations:
- Connectivity:
   - UDP
   - TCP
   - BlueTooth
   - Serial
- Packet format
  - Possibly different for different connectivity paradigms
- Streamlining and sterilizing the .ino
- making simple to moderate changes easy to make
- File and function naming and organization for ease of location and use
- Adherence to a uniform style guide (Possibly python)
   - Making all variables and functions public?
- Use with different feathers and boards
   - https://docs.google.com/spreadsheets/d/1bcQ6aMfaUTOqJm5eUgTE9E1hOmnVxvdWmdocrDOp8wg/edit#gid=0

#### Hardware Considerations:
- Use with different feathers and boards
- Cutting tracks
- Pins that can be repurposed
- Re-flowing or redirecting tracks:
  - https://www.eevblog.com/forum/manufacture/improved-protoboardperfboard-for-direct-soic-mounting/

produceconsumerobot, 2019-07-15:
[image]

badfishblues, 2019-08-19:
https://docs.google.com/drawings/d/1DuZusbsETLMmHD5LmeqF8rN3vBc_FgQ9w5PRoIj_5hs/edit