EmotiBit/EmotiBit_FeatherWing #92: Oscilloscope Momentarily greys out all EmotiBits

nitin710, 2020-03-02:
3 emotibits on network. all 3 initially shown as available on the oscilloscope.
 Established connection to emotibit3. data streams perfectly.
after some time, the emotibit 3 "blue" light turns OFF, and the emotibit is displayed as "greyed out" on the oscilloscope.Green light is still ON, Yellow light is not blinking, indicating emotibit is not sending control  channel messages.
the EmotiBIt red record light is still functional. Serial port is also responsive is debug mode.
but emotibit is still greyed out in the oecilloscope.
toggled the button. Wifi Mode turned OFF, then ON, [verified on the scope]
emotibit connects to Wifi, and the "OSCILLOSCOPE MOMENTARILY GREYS OUT ALL EMOTIBITS. then all emotibit's are shown to be available to connect.

produceconsumerobot, 2020-03-03:
@nitin710 I think this may be an issue when the network is dropping lots of packets and the advertising packets aren't getting through for the timeout period (~3 seconds as I recall).

nitin710, 2020-04-08:
@produceconsumerobot On analyzing the behaviour, it does seem like its when the network is slammed with traffic. Requesting feedback on whether to close this issue or not.

closed 2025-07-01 by nitin710