EmotiBit/EmotiBit_FeatherWing #154: FW v1.2.86 hangs when unplugged from EmotiBit

produceconsumerobot, 2021-09-27:
**Describe the bug**
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Unplug EmotiBit from Feather
2. Bootup Feather
3. Observe hung state trying to talk to sensor

**Expected behavior**
Expected behavior would be to hibernate

nitin710, 2021-09-27:
Since we switched from detecting SD-CArd to detecting pin voltage for version detection, https://github.com/EmotiBit/EmotiBit_FeatherWing/blob/c8025bbcd572c6c1be7c41df911a3e4a04967287/EmotiBitVersionController.cpp#L285
If no EmotiBIt is attached, i think the floating voltage is being detected as a false LOW. WHich is causing the FW to try and communicate with the SI-7013. 

But,  since the I2C lines are not pulled up, it is leading to I2C blocking the Execution

nitin710, 2022-01-17:
Fixed in 1.3+

closed 2022-01-17 by nitin710