EmotiBit/EmotiBit_FeatherWing #172: Add HR to EmotiBit FW

nitin710, 2022-01-27:
# Description
- Adding Heart Rate derivative metric to EmotiBit cpp

# Proposed changes
```
in processHeartRate()
{
// static variables
static int SAMPLE_BUFFER_CAPACITY = 2*samplingRate;
static int BEAT_LOCATION_BUFFER_CAPACITY = 10; // taking a heart rate of 180, that is 3 beats a second, and a size 10 can hold 6 beat locs
static float samples[SAMPLE_BUFFER_CAPACITY];
static int beatLocs[BEAT_LOCATION_BUFFER_CAPACITY];
uint8_t colorUsedForBeatDetection = TypeTag::IR; // maybe we can define another function as a setter to pipe in different data

// Step 1: Add new samples from IR/RED/GREEN dataBuffer to samples buffer
// this includes shifting all old values from End to the front and adding new values to the end

// Step 2: Adjust existing BeatLocs by the number of new samples
// each locValue is reduced by the #newSamples. if the shifted value is <0, that means the corresponding sample is out of the buffer. 
// Shift the entire beatLoc buffer so that only +ve values remain.

// Step 3: for all new samples, pass it through beat detector algo
// For every beat detected, add it to the beatLoc buffer

// Step 4: For all values in beatLocs, find avg interBeat sample interval

// Step 5: Use sampling rate to calculate the avg heart Rate.

// Step 6: Push calculated heart rate into HR dataDoubleBuffer

// Step 7: swap HR buffer
}
```

produceconsumerobot, 2022-01-27:
We should also send IBIs. These will be un-filtered vs `HR` which will likely have some kind of filtering on it.

produceconsumerobot, 2022-01-28:
We don't need anything calculated in the main loop to be double buffered -- only a single buffer is required. 
How are you thinking to manage this new derivative metric type of data, i.e. aperiodic data that's calculated in the main loop?

nitin710, 2022-02-02:
# Draft 2
## modifications from Draft 1
- removing all static buffers which may be used in the Sp)2 calculation.
- making it lean to just focus on IBI and HR

## Proposed changes to incorporate HR
```
in processHeartRate()
{
static uint32_t interBeatCount = 0;
static int basisSignal = DataType::IR; //  we can choose to remove it. Make HR only dependent on IR(conforms to datasheet)
static lowPassFilter; // digital IIR filter
float* data;
size_t dataLen;

// read basis signal for new samples
dataLen = dataDoubleBuffer[basisSignal]->getData(data,timestamp);
if(dataLen)
{
for(dataLen)
if(detectBeat(sample))
// is _outDataPacket of size 0? Check for outDataPacket size overflow
// add interbeat interval data to outDataPacket       

// calculate HR
timeBetweenBeat = interBeatCount/saplingrate.PPG;
HR = 60/timebetweenBeat;
HR = lowPass(HR);
// add HR into outdatapacket
interBeatCount = 0;
}
else
{
interBeatCount++;
}
}
```

```
in emotibit.cpp
in processData()
{
for(DataTypes)
if(datatype == HR)
processHeartRate()
}

nitin710, 2022-02-02:
@produceconsumerobot what do you think of draft 2?
Still trying to think of the best was to bypass `typeTag`

nitin710, 2022-02-03:
## Notes from discussion
- Represent IBI in mS
- Create a new signature for `addPacket()` that takes in `string` typetag
- IBI and HR should have decimal precision 0

nitin710, 2022-02-03:
# Draft 3

## Changes from draft 2
- Add a new signature for `addPacket`
- 

## Proposed changes to incorporate HR
```
in processHeartRate()
{
static uint16_t interBeatSampleCount = 0;
static int basisSignal = DataType::IR; //  we can choose to remove it. Make HR only dependent on IR(conforms to datasheet)
static heartRateFilter; // digital IIR filter @ some low frequency. maybe 2 Hz?
float* data;
size_t dataLen;
const uint8 precision = 0;
static const float timePeriod = 1/ sampling rate;

// read basis signal for new samples
dataLen = dataDoubleBuffer[basisSignal]->getData(data,timestamp);
if(dataLen)
{
for(dataLen)
  interBeatSampleCount ++;
if(detectBeat(sample))
  // add interbeat interval data to outDataPacket       
  // signature: addPacket(uint32_t timestamp, EmotiBit::DataType t, float * data, size_t dataLen, uint8_t precision). // default precision is 4

  // calculate HR
  timeBetweenBeat = interBeatSampleCount * timePeriod ;
  HR = 60/timebetweenBeat;
  HR = heartRateFilter(HR);
  // add HR into outdatapacket
  interBeatCount = 0;
}
}
```

```
in emotibit.cpp
in processData()
{
for(DataTypes)
if(datatype == HR)
processHeartRate()
}

// new
void addPacket(uint32_t timestamp, const char* typeTag, float * data, size_t dataLen, uint8_t precision = 0)
{
  protocolversion = 1;
  String headerString = getPacketHeader(const char* typeTag, timestamp, dataLen, protocolversion);
  _outDataPackets += headerString;
 String dataString = getDataString(data, dataLen);
  _outDataPackets += dataString;
}

// new
String getPacketHeader(const char* typeTag, timestamp, dataLen, protocolversion);
{
  header = EmotiBitPacket::createHeader(typeTags[(uint8_t)t], timestamp, _outDataPacketCounter++, dataLen, protocolVersion);
  String headerString = EmotiBitPacket::headerToString(header);
  return headerString ;
}

// new
void get PacketData(float* data, int dataLen)
{
  String dataString;
  for (uint16_t d = 0; d < dataLen; d++) {
    String temp = EmotiBitPacket::PAYLOAD_DELIMITER + String(data[d], precision);
    if (_outDataPackets.length() + temp.length() < OUT_MESSAGE_RESERVE_SIZE - 1)
    {
       dataString+= temp;
    } 
    else
    {
      dataString+= EmotiBitPacket::PAYLOAD_TRUNCATED;
      Serial.print("ERROR: _outDataPackets exceeded OUT_MESSAGE_RESERVE_SIZE ");
      Serial.println("[" + String(_outDataPackets.length()) + "+" + String(temp.length()) + "/" + String(OUT_MESSAGE_RESERVE_SIZE + "]"));
      break;
    }
  }
  return dataString;
}

```

nitin710, 2022-02-22:
# Testing 

- below, `U1` is a filtered IR intermediary.
- HR is the detected heart rate
[image]

### Code snippet to send an intermediary to the oscilloscope
```
float filteredPpg[10];
	if (bufferSize < 10 && bufferSize > 0)
	{
		for (int i = 0; i < bufferSize; i++)
		{
			filteredPpg[i] = ppgSensorHighpass.filter(bufferData[i]);
			filteredPpg[i] = ppgSensorLowpass.filter(filteredPpg[i]);
		}
		addPacket(timestamp, "U1", filteredPpg, bufferSize);
	}
```

closed 2022-03-05 by nitin710