EmotiBit/EmotiBit_FeatherWing #173: Add EDR Amplitude and EDR frequency to EmotiBit stock

nitin710, 2022-02-02:
# Description
- Add data type record EDR events
  - An Electrodermal response is a physiological activity, which measures and logs a sginificant change in EDA over a short period of time. 
  - External stimulus can generate a physiological response, and this response can be logged using EDR

nitin710, 2022-02-02:
# Draft 1

## Proposed changes
```
in processElectrodermalResponse()
{
// static variables
float* data;
size_T dataLen;
highPassFilter // digital Filter
lowPassFilter; // to filter out the EDR frequency
float filteredSignal;
static float threshold; // is this not dependent on the base EDA? should this be a table?
static int interResposeSampleCount; // to count number of samples passed between EDR events
static interResponseTime;

// Load latest EDA signal
dataLen = dataDoubleBuffer[EDA]->getData();

// process EDA
for(dataLen)
{
filteredSignal = highPassFilter(data[i]);
if(filtererdSignal > threshold)

// add packet for EDR response. stored 
// do we just store a millis() as a time stamp?

// calculate the time between response
interResponseTime = lowPassFilter( interResponseSampleCOunt/samplingFreq);

// Add packet for EDR frequency

}
```

nitin710, 2022-02-02:
@produceconsumerobot what do you think of the draft? Still trying to think of a way to bypass `typeTag`

nitin710, 2022-02-03:
## Notes from Discussion
- Send `rise time` as well
- EDR Amplitude is the peak of the response
- EDR Frequency depends on the time-onset of a response
- On threshold crossing
  - we store the time.
  - flip a bool
  - update and send the freq
  - in the bool subState
    -  we wait for a drop on the EDA
    - rise time = sample till peak - onset
    - amplitude = peak 
    - send both
    - flip the bool back

nitin710, 2022-02-03:
# Draft 2
```
in processElectrodermalResponse()
{
float* data;
size_T dataLen;
edrAmplitudeHighPassFilter // for bandpass @ 0.5Hz
edaAmplitudeLowPassFilter // for bandpass @1.5Hz
edaAmplitudeBaselineFilter // @0.1Hz  // used to provide a baseline that works as a reference for threshold
edrFrequencyFilter // lowPass digital filter @1Hz
static float threshold; // ~ 0.1 (10%)
static uint16 interResposeSampleCount; // to count number of samples passed between EDR events
float interResponseTime;
static const float timePeriod = 1/ samplingFreq;
static bool onsetDetected;
static float baseEdrAmplitude;
float riseTime;

// Load latest EDA signal
dataLen = dataDoubleBuffer[EDA]->getData();

// process EDA
for(dataLen)
interResponseSampleCount++;
{
  lowPassfilteredSignal = edaAmplitudeLowPassFilter (data[i]);
  highPassfilteredSignal = edrAmplitudeHighPassFilter(lowPassfilteredSignal); // creates a bandpass by cascading low and high
  if(!onsetDetected)
  {
    if( highPassfilteredSignal / lowPassfilteredSignal > threshold)
    {
      // set onset detected.
      // record onset time(by back-calculating based on frequency)
      // set base edr amplitude
      baseEdrAmplitude = data[i]
    
      // calculate the time between response
      interResponseTime = lowPassFilter( interResponseSampleCount * timePeriod);
      // Add interResponseTime  _outDataPackets
      // interResponseSampleCount  = 0;
    }
  }
  else  // after onset has been detected
  {
    if(data[i] < data[i-1]) // peak detected
    {
         amplitude = data[i-1] - baseEdrAmplitude;
    }
    // calculate the rise time
   riseTime = interResponseSampleCount  * timePeriod;
   // send TimePeriod
   // onsetDetected = false; 
  }
}
```

produceconsumerobot, 2022-02-24:
[image]
- from Boucsein et al 2021

produceconsumerobot, 2022-02-24:
SCR duration is also a commonly used term, e.g. from Handler et al. 2012
An EDA Primer for Polygraph Examiners
[image]

produceconsumerobot, 2022-02-24:
from Setz et al 2010
[image]

produceconsumerobot, 2022-02-24:
From Boucsein's book 
[image]
[image]

nitin710, 2022-02-25:
A Guide for Analysing Electrodermal Activity (EDA) & Skin Conductance Responses (SCRs) for Psychological Experiments
https://www.birmingham.ac.uk/Documents/college-les/psych/saal/guide-electrodermal-activity.pdf

nitin710, 2022-02-25:
@produceconsumerobot as discussed, we may want to move the `EDR:FREQ` to a periodic channel. Let's pool that discussion into this issue.

Early draft:
[image]

nitin710, 2022-02-28:
Moving processElectrodermalActivity() into `EmotiBitEda`
- sampling rate exists in the `EmotiBitEda()` class
- `dataDoubleBuffers[(uint8_t)EmotiBit::DataType::EDA]` is replaced with the class private member `DoubleBufferFloat`

### Proposed solution for `addPacket`
- AddPacket() needs
  - timestamp
  - typetag
  - data
  - datalen
- We can move all the mathematics inside `EmotiBitEda`
- in EmotiBit.cpp
  - in processAperiodic()
    - onsetTime = emotiBitEda.getOnsetTime();
    - responseFreq = emotiBitEda.getResponseFreq();
    - dataLen = emotiBitEda.getDataLen();
    - addPacket(onsetTime, EmotiBitPacket::TypeTag::ELECTRODERMAL_RESPONSE_FREQ, &responseFreq, dataLen, 4); // 4 = precision

nitin710, 2022-03-04:
## Typetag Nomenclature

**possible typetags between `E` series and `S` series**
|Signal name | Typetag(S series) | Typetag(E series)|
|--------------|--------------------|-------------------|
|EDA|EA|EA|
|EDL|EL|EL|
|EDR|ER|ER|
|Skin Conductance Amplitude|SA|EC(electrodermal change)|
|Skin Conductance Response Frequency|SF|EF|
|Skin Conductance Response Rise Time|ST|ET|
|Skin Conductance Response Recovery|SR|E?|
|Skin Conductance Response Duration|SD|ED|
|Skin Conductance Response Latency|SL|E?|

### S- series
**Reasons to have S series**
- The literature seems to annotate this physiological phenomenon as "Skin conductance response", so it feels like a fit
- We are also reporting every data in uS, which is conductance
- Electrodermal activity (EDA) is the umbrella term used for defining autonomic changes in the electrical properties of the skin. The most widely studied property is the skin conductance, [link](https://www.birmingham.ac.uk/Documents/college-les/psych/saal/guide-electrodermal-activity.pdf)
- if people are interested in Resistance, they may use conductance and derive resistance in post-processing.

**note:** when we add Skin conductance duration, need to resolve conflict with type tag SD-Card percentage.

** Reasons not to have S series**
- Conflict with typetags in Factory test, but we can keep them isolated by enforcing serial(setup related) and wifi comms(data related) to be separate.

### R-series

**Reasons to Have E series**
- Legacy
  - We have had EDA EDR and EDL before
- E for Electrodermal has an intuitive meaning

**Reasons not to have it**
- Outstanding namespace conflicts
  - Recovery
  -  Latency
- Electrodermal Activity is an umbrella for various signal, and we are trying to represent, specifically, skin conductance phenomenons
  - SCRs are a subset of EDA representation
  - Using SCRs may make sense in trying to zone in the meaning of the signal

nitin710, 2022-03-05:
Here `U1` represents a bandpass filtered eda, which is being used to determine the SCR derivatives. bandpass **[0.3-1] Hz**
[image]

closed 2022-03-05 by nitin710