EmotiBit/EmotiBit_FeatherWing #202: Fix ESP32 _dataFile.sync()

produceconsumerobot, 2022-06-28:
We either need to utilize SdFat library for ESP32 or find/create similar functionality for built-in SD library

nitin710, 2022-06-28:
Ok, i shall look into it!

Aslo, additional information in the issue description (like links to code segments) is always welcome!

produceconsumerobot, 2022-06-28:
see also https://github.com/greiman/SdFat/issues/323

produceconsumerobot, 2022-06-28:
If it's not reasonably easy to avoid SdFat conflicts on ESP, using _dataFile.flush() on ESP322 instead of _dataFile.sync() fixes the problem. It may be possible to only use .flush(), i.e. with both standard SD.h and SdFat.h, though possibly not worth the engineering to assess if there are any issues on M0, so a ToDo to assess may be the best agile path.

PR #207 (2022-07-13): Feat esp isr opt
# Features
- ESP32 is basically working
- CPU 80MHz
- ISR 150Hz on core 1
- Added ISR readSensorsInterval readSensorsDurationMax readout
- Fixed thermopile timestamps

# Issues Referenced
- closes #202 

# Tests Passed
- ✔️ readSensorsInterval +/- ~100 uSec
- ✔️ readSensorsDurationMax theoretical 5317 uSec, empirical 5196 uSec, 48% CPU
- ✔️ Timesync RTs ~4 mSec
- ✔️ Full streaming battery life ~3.9 hours
- ✔️ timestamp_check.py
- ✔️ no DOs / DCs