EmotiBit/EmotiBit_FeatherWing #282: feather freezes randomnly

devriesewouter89, 2023-09-07:
**Describe the bug**
After a time being connected and recording data, the feather suddenly stops recording and freezes. This also happens sometimes when stopping and restarting a recording. 
We have two setups running in parallel (each time a recording laptop + Emotibit + router) for a research project. 
With serial via USB connected the feather doesn't respond as well anymore, blue light stays on, red light stops flashing (goes to off).

**Additional details**
- EmotiBit software version: 1.9.0
- EmotiBit firmware version: 1.9.0 
- EmotiBit hardware version: V04a
- OS version: windows 10

**To Reproduce**
Steps to reproduce the behavior:
I'll detail only the restarting of recording behavior, yet it also happens at random times, when no command whatsoever is given
1. Go to Emotibit oscilloscope
2. Click on recording, wait a second, stop the recording and restart recording quickly after.
3. Emotibit oscilloscope: The Emotibit device becomes grayed out in the device list.
4. Emotibit device: The Emotibit itself does not respond anymore via terminal, blue light is continuously on (not flashing!), red light stops.

Ghazalchizarian, 2024-10-20:
Hi, were you ever able to fix this problem? I have the exact same issue

devriesewouter89, 2024-10-21:
Hi
I had to change the microcontroller, if I recall to and esp32 based one,
since I'm living in the EU sending the microcontroller for debugging was
too costly, so we dont know the origin of the error
Sincerely

nitin710, 2024-10-22:
@Ghazalchizarian are you using the EmotiBit with a Feather M0? If yes, can you upgrade to the [latest fw](https://github.com/EmotiBit/EmotiBit_FeatherWing/releases/tag/v1.12.1)? There was a fw patch that improved stability.

closed 2025-07-01 by nitin710