EmotiBit/EmotiBit_FeatherWing #313: Unable to successfully complete PlatformIO build

martaah7, 2024-07-01:
**Describe the bug**
Build keeps failing, detailed [here](https://www.reddit.com/r/EmotiBit/comments/1dodpy5/platformio_build/).

**Expected behavior**
Getting Status: FAILED, expecting a success

**Additional details**
Error log:
`Executing task: C:\Users\NDNEL\.platformio\penv\Scripts\platformio.exe run --environment adafruit_feather_m0 

Warning! Ignore unknown configuration option `firmware_name_board_name` in section [env:adafruit_feather_m0]
Warning! Ignore unknown configuration option `firmware_name_board_name` in section [env:adafruit_feather_esp32]
Processing adafruit_feather_m0 (platform: atmelsam @3.8.1; board: adafruit_feather_m0; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/adafruit_feather_m0.html
PLATFORM: Atmel SAM (3.8.1) > Adafruit Feather M0
HARDWARE: SAMD21G18A 48MHz, 32KB RAM, 256KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink)
PACKAGES:
 - framework-arduinosam @ 4.3.190711
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
Converting EmotiBit_stock_firmware.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- EmotiBit FeatherWing @ 1.11.1
|-- Wire @ 1.0
|-- EmotiBit SI7013 @ 0.0.7
|-- EmotiBit BMI160 @ 0.3.3
|-- EmotiBit MAX30101 @ 2.0.3
|-- EmotiBit NCP5623 @ 0.0.4
|-- EmotiBit MLX90632 @ 1.0.8
|-- ArduinoJson @ 6.21.2
|-- SPI @ 1.0
|-- Adafruit Zero DMA Library @ 1.0.4
|-- WiFi101 @ 0.16.1
|-- EmotiBit XPlat Utils @ 1.6.0
|-- SdFat @ 2.2.3
|-- Arduino Low Power @ 1.2.2
|-- RTCZero @ 1.6.0
|-- SAMD_AnalogCorrection @ 1.0
|-- EmotiBit External EEPROM @ 1.0.5+EmotiBit.0.0.1
|-- EmotiBit ADS1X15 @ 2.2.0+EmotiBit.0.0.1
|-- Adafruit BusIO @ 1.16.1
|-- SimpleFTPServer @ 2.1.7
Building in release mode
Compiling .pio\build\adafruit_feather_m0\src\EmotiBit_stock_firmware.ino.cpp.o
Compiling .pio\build\adafruit_feather_m0\libce8\EmotiBit_SimpleFTPServer\FtpServer.cpp.o
C:\Users\NDNEL\Documents\Arduino\libraries\EmotiBit_SimpleFTPServer\FtpServer.cpp: In member function 'bool FtpServer::processCommand()':
C:\Users\NDNEL\Documents\Arduino\libraries\EmotiBit_SimpleFTPServer\FtpServer.cpp:675:77: error: use of deleted function 'FsFile::FsFile(const FsFile&)'
             client.print( F(";Size=") ); client.print( long( fileSize( file )) );
                                                                             ^
In file included from C:\Users\NDNEL\Documents\Arduino\libraries\SdFat\src/FsLib/FsLib.h:31:0,
                 from C:\Users\NDNEL\Documents\Arduino\libraries\SdFat\src/SdFat.h:33,
                 from C:\Users\NDNEL\Documents\Arduino\libraries\EmotiBit_SimpleFTPServer/FtpServer.h:40,
                 from C:\Users\NDNEL\Documents\Arduino\libraries\EmotiBit_SimpleFTPServer\FtpServer.cpp:49:
C:\Users\NDNEL\Documents\Arduino\libraries\SdFat\src/FsLib/FsFile.h:918:7: note: 'FsFile::FsFile(const FsFile&)' is implicitly deleted because the default definition would be 
ill-formed:
 class FsFile : public StreamFile<FsBaseFile, uint64_t> {
       ^~~~~~
C:\Users\NDNEL\Documents\Arduino\libraries\SdFat\src/FsLib/FsFile.h:918:7: error: use of deleted function 'StreamFile<FsBaseFile, long long unsigned int>::StreamFile(const StreamFile<FsBaseFile, long long unsigned int>&)'
In file included from C:\Users\NDNEL\Documents\Arduino\libraries\SdFat\src/ExFatLib/ExFatFile.h:883:0,
                 from C:\Users\NDNEL\Documents\Arduino\libraries\SdFat\src/ExFatLib/ExFatVolume.h:27,
                 from C:\Users\NDNEL\Documents\Arduino\libraries\SdFat\src/ExFatLib/ExFatLib.h:28,
                 from C:\Users\NDNEL\Documents\Arduino\libraries\SdFat\src/SdFat.h:31,
                 from C:\Users\NDNEL\Documents\Arduino\libraries\EmotiBit_SimpleFTPServer/FtpServer.h:40,
                 from C:\Users\NDNEL\Documents\Arduino\libraries\EmotiBit_SimpleFTPServer\FtpServer.cpp:49:
c:\users\ndnel\documents\arduino\libraries\sdfat\src\common\arduinofiles.h:61:7: note: 'StreamFile<FsBaseFile, long long unsigned int>::StreamFile(const StreamFile<FsBaseFile, long long unsigned int>&)' is implicitly deleted because the default definition would be ill-formed:
[... 43 lines cut ...]
C:\Users\NDNEL\Documents\Arduino\libraries\EmotiBit_SimpleFTPServer\FtpServer.cpp: In member function 'bool FtpServer::doMlsd()':
C:\Users\NDNEL\Documents\Arduino\libraries\EmotiBit_SimpleFTPServer\FtpServer.cpp:1691:65: error: use of deleted function 'FsFile::FsFile(const FsFile&)'
     data.print( F(";Size=") ); data.print( long( fileSize( file )) ); data.print( F("; ") );
                                                                 ^
In file included from C:\Users\NDNEL\Documents\Arduino\libraries\EmotiBit_SimpleFTPServer\FtpServer.cpp:49:0:
C:\Users\NDNEL\Documents\Arduino\libraries\EmotiBit_SimpleFTPServer/FtpServer.h:664:12: note:   initializing argument 1 of 'uint32_t FtpServer::fileSize(FsFile)'
   uint32_t fileSize( FTP_FILE file );
            ^~~~~~~~
Compiling .pio\build\adafruit_feather_m0\lib738\EmotiBit_FeatherWing\EmotiBitConfigManager.cpp.o
*** [.pio\build\adafruit_feather_m0\libce8\EmotiBit_SimpleFTPServer\FtpServer.cpp.o] Error 1
C:\Users\NDNEL\Documents\Arduino\libraries\EmotiBit_FeatherWing\EmotiBitConfigManager.cpp: In member function 'void EmotiBitConfigManager::updateWiFiCredentials(String, String, uint8_t)':
C:\Users\NDNEL\Documents\Arduino\libraries\EmotiBit_FeatherWing\EmotiBitConfigManager.cpp:250:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
        if(deleteIndex < configAsJson["WifiCredentials"].size())
           ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\NDNEL\Documents\Arduino\libraries\EmotiBit_FeatherWing\EmotiBitConfigManager.cpp:297:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       for(int i = 0; i < configAsJson["WifiCredentials"].size(); i++ )
                      ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\NDNEL\Documents\Arduino\libraries\EmotiBit_FeatherWing\EmotiBitConfigManager.cpp:108:30: warning: unused variable 'fileParses' [-Wunused-variable]
     bool fileExists = false, fileParses = false;
                              ^~~~~~~~~~
========================================================================= [FAILED] Took 31.82 seconds =========================================================================
Environment          Status    Duration
-------------------  --------  ------------
adafruit_feather_m0  FAILED    00:00:31.824
==================================================================== 1 failed, 0 succeeded in 00:00:31.824 ====================================================================`

nitin710, 2024-07-01:
Can you downgrade your SdFat library to `2.2.0`? Our latest build was tested on that.

martaah7, 2024-07-02:
That fixed it. Thank you!

nitin710, 2024-07-02:
Awesome. Looks like the latest SdFat has deprecated something.

I [updated the documentation](https://github.com/EmotiBit/EmotiBit_Docs/commit/1ff4d62294245a9f81a9a381613e5bc23a26ef4b) as well.

closed 2024-07-02 by nitin710